LivotovLabs/CamView #13: executing error

carlosnavas91, 2015-09-10:
Hi, how are you... I'm trying this library and first I had a problem with the AppCompat library, so I declared it in the dependencies as follows:

compile ('eu.livotov.labs:camview:2.0.0.DEV1-SNAPSHOT@aar') {
        transitive=true
        exclude module:'appcompat-v7'
    }

And it worked fine, so I continued with the coding and when it's time to execute the program it throws the following message:

Error:Execution failed for task ':app:dexDebug'.
> com.android.ide.common.process.ProcessException: org.gradle.process.internal.ExecException: Process 'command 'C:\Program Files\Java\jdk1.7.0_79\bin\java.exe'' finished with non-zero exit value 2

I assume that's a problem with libraries and incompatibilities... So what should I do?

thanks :)

livotov, 2015-09-10:
Hi !

The library is build with api 23, so maybe this is the cause ? What build tools and api's you're using in your app ?

Here is the build file of my test app which uses CamView:

```
buildscript {
    repositories {
        mavenCentral()
    }
    dependencies {
        classpath 'com.android.tools.build:gradle:1.3.0'
    }
}
apply plugin: 'com.android.application'

repositories {
    mavenCentral()
}

android {
    compileSdkVersion 23
    buildToolsVersion "22.0.1"

    defaultConfig {
        minSdkVersion 14
        targetSdkVersion 23
        versionCode 1
        versionName "1.0"
    }
    buildTypes {
        release {
            minifyEnabled false
            proguardFiles getDefaultProguardFile('proguard-android.txt'), 'proguard-rules.txt'
        }
    }
}

dependencies {
    compile fileTree(dir: 'libs', include: ['*.jar'])
    compile ('eu.livotov.labs:camview:2.0.0.DEV1-SNAPSHOT@aar') {transitive=true}
    compile 'com.android.support:appcompat-v7:23.0.0'
}
```

carlosnavas91, 2015-09-10:
Look, this is how I have my gradle file:

buildscript {
    repositories {
        maven { url 'https://maven.fabric.io/public' }
    }

    dependencies {
        classpath 'io.fabric.tools:gradle:1.+'
    }
}
apply plugin: 'com.android.application'
apply plugin: 'io.fabric'

android {

    signingConfigs {
        debug {
            storeFile file('../keystores/debug.keystore')
        }
    }
    compileSdkVersion 22
    buildToolsVersion "22.0.1"
    defaultConfig {
        applicationId "and.doo.greenggers.com.doo"
        minSdkVersion 15
        targetSdkVersion 22
        versionCode 1
        versionName "1.0"
    }
    buildTypes {
        release {
            minifyEnabled false
            proguardFiles getDefaultProguardFile('proguard-android.txt'), 'proguard-rules.pro'
        }
    }
}

repositories {
    mavenCentral()
    maven { url 'https://maven.fabric.io/public' }
    maven { url 'http://maven.livotovlabs.pro/content/groups/public' }
}

dependencies {
    compile fileTree(dir: 'libs', include: ['*.jar'])
    compile files('libs/httpmime-4.2.6.jar')
    compile files('libs/gcm.jar')

    compile('com.google.android.support:wearable:1.1.0')
    compile('com.google.android.gms:play-services:7.5.0') {
        exclude module: 'support-v4'
    }
    compile('com.android.support:appcompat-v7:22.1.0')

    compile('com.twitter.sdk.android:twitter:1.5.1@aar') {
        transitive = true;
    }

    compile 'com.squareup.okhttp:okhttp:2.4.0'

    compile 'com.google.maps.android:android-maps-utils:0.3+'
    compile 'com.squareup.retrofit:retrofit:1.9.0'

    compile 'com.facebook.android:facebook-android-sdk:4.1.2'
    compile 'com.github.navasmdc:MaterialDesign:1.5@aar'
    compile 'com.ogaclejapan.arclayout:library:1.0.1@aar'
    compile('com.nineoldandroids:library:2.4.+')

    compile ('eu.livotov.labs:camview:2.0.0.DEV1-SNAPSHOT@aar') {
        transitive=true
        exclude module:'appcompat-v7'
    }

}

carlosnavas91, 2015-09-10:
I builded it just like you are doing it, and is showing the same message :/ 

please help :+1:

livotov, 2015-09-11:
did you set compileSdkVersion 23 for your project ?

livotov, 2015-09-11:
does it compile without CamView ?
because I see you're including jar files from your libs folder twice

```
compile fileTree(dir: 'libs', include: ['*.jar'])
compile files('libs/httpmime-4.2.6.jar')
compile files('libs/gcm.jar')
```

this may cause dex error too (duplicate classes). Try remove duplicate imports and leave only generic import statement:

```
compile fileTree(dir: 'libs', include: ['*.jar'])
```

carlosnavas91, 2015-09-11:
Yes, I did set the compileSdk to 23 and still doesn't work... if I remove the:

     compile ('eu.livotov.labs:camview:2.0.0.DEV1-SNAPSHOT@aar')

The project compile without problem, but indeed the error is about "dexDebug"
I also try by only leaving the generic import statement: 
     compile fileTree(dir: 'libs', include: ['*.jar'])

And it still doesn't work.....

What else could it be?

livotov, 2015-09-11:
hm, could you please grab more build logs or run the gradle from console, I believe there should be more error messages before the final "dexDebug error"

livotov, 2015-09-16:
so, how it is ?

carlosnavas91, 2015-09-16:
Dmitri how are you.... well finally I'm using this library
https://github.com/dm77/barcodescanner which I'm not to happy about, but it
is not being a problem to me... about the log messages I could import it
again and then send it to you, but as far as I know that was a problem with
duplicated libraries and I tried excluding all the modules I thought could
be generating those incompatibilities; I also tried what you said about
compiling with the 23 SDK version and building with the 22.0.1 version.

Carlos Navas
Ingeniero Multimedia
Skype: navas.carlos91
(+57) 318 523 2287
co.linkedin.com/in/carlosenavas

livotov, 2015-09-16:
well, try with 2.0.0.DEV3-SNAPSHOT - I removed dependency on appcompat in the library, so maybe this will help in your installation.

mtangoo, 2015-09-17:
Any reason for code  in `  exclude module:'appcompat-v7'`

```
 compile ('eu.livotov.labs:camview:2.0.0.DEV1-SNAPSHOT@aar') {
    transitive=true
    exclude module:'appcompat-v7'
  }

livotov, 2015-09-17:
Since DEV3 snapshot there is no more dependency on appcompat in camview.

closed 2015-10-06 by livotov