LivotovLabs/CamView #9: Installing on Maven Latest

mtangoo, 2015-09-10:
I want to install 2.0.0.DEV1-SNAPSHOT (latest). Putting in version does not work.
Would you update the installation guide (telling us what to put in "version" in gradle, instead of x.y.z) for getting the latest?

I'm sorry that am not Gradle/Maven maverick if this is obvious!
S.

livotov, 2015-09-10:
Simply replace x.y.z to 2.0.0.DEV1-SNAPSHOT 
and that's all ;)

mtangoo, 2015-09-10:
it did not work. Did you try it? Grale complains that library is not found
I will try again and see if there is any change.

I use Android studio latest

livotov, 2015-09-10:
did you include the maven repo url as well ? Because it is not in the maven central but on our own company repo, so you have to add the repo url as described in the readme.

Anyway, updated readme to replace x.y.z to latest version :)

mtangoo, 2015-09-10:
I was doing it wrong way. Thanks for docs!

closed 2015-09-10 by mtangoo