LivotovLabs/CamView #14: in old version i can`t stoped(to release the camera) the camera

Y-E-P, 2015-09-11:
I can`t stop the camera for free a camera process for other applications.

    @Override
    public void onBackPressed() {
        scanner.stopScanner();
        super.onBackPressed();

    }

    @Override
    protected void onPause() {
        scanner.stopScanner();
        super.onPause();
    }

    /*@Override
    protected void onStop() {
        scanner.stopScanner();
        super.onStop();
    }*/

    @Override
    protected void onResume() {
        scanner.startScanner();
        super.onResume();
    }

    @Override
    protected void onDestroy() {
        scanner.stopScanner();
        scanner = null;
        super.onDestroy();
    }

livotov, 2015-09-11:
Sorry, did not get what you mean. You mean that stopScanner() method in 2.0.0.DEV1-SNAPSHOT does not release the camera ? Or something other ?

Y-E-P, 2015-09-11:
When I press a "Home button" to hide the my application and open other app(wich uses a camera) i take a error(other program can use a camera because scanner.stopScanner() is not released a camera).

I use this library compile('eu.livotov.labs:zxscanlib:2.0.1@aar') { transitive = true }

Y-E-P, 2015-09-11:
But on 2.0.0.DEV1-SNAPSHOT is works fine in my opinion:)

livotov, 2015-09-11:
Yes, this seems to be a bug in previous version. Also checked 2.0.0 and it releases the camera correctly on stopScanner

closed 2015-09-11 by livotov

Y-E-P, 2015-09-11:
like this compile('eu.livotov.labs:zxscanlib:2.0.0@aar') { transitive = true }

livotov, 2015-09-11:
no, like that:

```
 compile ('eu.livotov.labs:camview:2.0.0.DEV1-SNAPSHOT@aar') {transitive=true}
```

Note, that zxscanlib project is deprecated now. All further development is performed here, in CamView repo.

Y-E-P, 2015-09-11:
this library is not work correctly,i have this problem https://github.com/LivotovLabs/CamView/issues/7

livotov, 2015-09-11:
#7 was copied from zxscanlib. It contains the same problem which is not fixed at the moment

Y-E-P, 2015-09-11:
but on old zxscanlib camera works fine for me...

Y-E-P, 2015-09-11:
ok,i use old fersion.but how i can release() a camera???

livotov, 2015-09-11:
basically it should release it - it calls the proper method

```
public void stopScanner()
    {
        shutodownThreadingSubsystem();
        camera.stop();
    }
```

so not sure why it is failing to release. you can checkout the sources and try to debug it

regarding the "but on old zxscanlib camera works fine for me..." - could you please create a new ticket for that and attach the screenshots with the problem ? Probably you're experiencing another issue.