LivotovLabs/CamView #15: Size of Scanning Area

mtangoo, 2015-09-15:
Currently scanning area is very small (red rectangle) and it makes hard to scan and adding time for recise scanning. I tried my a agains (This Ap)[https://play.google.com/store/apps/details?id=me.scan.android.client] and mine took time to put QR code right under the scanner.

Is there a way to increase (or at least provide a way) to make something like linked in app?

Y-E-P, 2015-09-15:
I think you must to set custom HUD setHudImageResource(R.drawable.my_hud_image);
it works for me

livotov, 2015-09-16:
@mtangoo , actually, red rectangle is just an artwork. Actual scanning area equals entire image d decoder reads entire preview from the camera (0,0 width, height), so you don't have to put your barcode right below the red rectrange. Even if you put your barcode to the corner, it will be decoded.

As @Y-E-P mentioned, you may put your own image via  setHudImageResource, however this will change just a visual appearance.

If it does not scan the qr code when in not exactly at the center, let me know, this could be some bug

P.S. don't forget there is a throttling timeout set for duplicate barcode, so once barcode is read, same one will not be decoded within some time (5 seconds by default, AFAIK).

mtangoo, 2015-09-16:
You are right. I had false impression that it somehow works only if you put it in the rectangle.
The real issue is I forgot to add Camera permission to Manifest. I thing it will do good to add it in readme

mtangoo, 2015-09-16:
@Y-E-P do you have sample Image

closed 2015-09-16 by mtangoo