LivotovLabs/CamView #16: Preserve Aspect Ratio

mtangoo, 2015-09-16:
I need to have Scanner width of let say 200dp but I want it to calculate height depending on the aspect ratio of the camera. Does CamView provide for that? I think something like resize(w, h) where if w or h is -1 you would resize preserving aspect ratio

livotov, 2015-09-16:
Initially it was designed that you can specify exact W & H as you want, and the camera manager will choose the appropriate preview size, enlarging the preview (in case when aspect ratio does no match) picture to fill empty spaces (yes, this will leave portion of the picture out of bounds but this is OK for code scanning).

For some reason, the preview picture aspect ration is broken at the moment. This what Im trying to fix

livotov, 2015-09-16:
This is the part of #7

mtangoo, 2015-09-16:
I will leave this open until #7 is closed

mtangoo, 2015-09-16:
[This SO post](http://stackoverflow.com/questions/17804309/android-camera-preview-wrong-aspect-ratio) seems to have information on same problem. I currently don't have enough time to look at it. but If I get any, try to fix it.

livotov, 2015-09-16:
Thanks !
pull requests are always welcome :)

livotov, 2015-09-16:
You can try sources from master or check the maven version 2.0.0.DEV2-SNAPSHOT
It contains quick fix attempt to aspect ratio. please try and let me know how it will work on your device

Thanks

mtangoo, 2015-09-16:
Any easy way to force android studio to update?
Nothing have changed so far 
(Ubuntu 15.04)

mtangoo, 2015-09-16:
Ahaa! I didn't see the 2. Ignore it!

mtangoo, 2015-09-16:
So I have tested but its not working normally.
But this is how to make it work (Note start and stop are my buttons which calls scanner.startScanner() and stopScanner() respectively):
start -- Does not resize
stop -- Does not resize
start -- Resizes fine

livotov, 2015-09-16:
Could you please show me your layout ?
Because I can't reproduce this at my side - whatever ScannerLiveView W & H I set, it resizes camera preview correctly.

mtangoo, 2015-09-16:
```xml
<RelativeLayout xmlns:android="http://schemas.android.com/apk/res/android"
    xmlns:tools="http://schemas.android.com/tools" android:layout_width="match_parent"
    android:layout_height="match_parent" android:paddingLeft="@dimen/activity_horizontal_margin"
    android:paddingRight="@dimen/activity_horizontal_margin"
    android:paddingTop="@dimen/activity_vertical_margin"
    android:paddingBottom="@dimen/activity_vertical_margin"
    tools:context="tz.co.hosannahighertech.ticketmanager.ScanMobileActivity">

    <eu.livotov.labs.android.camview.ScannerLiveView
        android:id="@+id/scannerView"
        android:layout_width="400dp"
        android:layout_height="200dp"
        android:layout_alignParentTop="true"
        android:layout_centerHorizontal="true" />

    <Button
        android:layout_width="match_parent"
        android:layout_height="wrap_content"
        android:text="Stop"
        android:id="@+id/stopButton"
        android:layout_above="@+id/scannerButton"
        android:layout_centerHorizontal="true" />

    <Button
        android:layout_width="match_parent"
        android:layout_height="wrap_content"
        android:text="Start"
        android:id="@+id/scannerButton"
        android:layout_alignParentBottom="true"
        android:layout_alignRight="@+id/stopButton" />

    <LinearLayout
        android:orientation="horizontal"
        android:layout_width="match_parent"
        android:layout_height="wrap_content"
        android:weightSum="1"
        android:id="@+id/linearLayout"
        android:layout_below="@+id/scannerView"
        android:padding="5dp">

        <ImageView
            android:layout_width="wrap_content"
            android:layout_height="wrap_content"
            android:paddingRight="10dp"
            android:id="@+id/imgMsg" />

        <TextView
            android:layout_width="match_parent"
            android:layout_height="match_parent"
            android:gravity="center_vertical"
            android:id="@+id/message"
            android:layout_weight="0.50"
            android:textStyle="bold"
            android:textSize="30sp"
            android:paddingRight="10dp" />

    </LinearLayout>

    <TextView
        android:layout_width="match_parent"
        android:layout_height="wrap_content"
        android:textSize="20sp"
        android:id="@+id/details"
        android:layout_alignParentRight="true"
        android:layout_alignParentEnd="true"
        android:layout_above="@+id/stopButton"
        android:layout_below="@+id/linearLayout" />

</RelativeLayout>

```

mtangoo, 2015-09-16:
Do you set size with XML or programatically?

livotov, 2015-09-16:
Yep, in XML.
Just copied your layout above to my test app, looks like everything is ok, see the video: http://share.livotov.eu/hlSe/p8TMlSNC

Could you please show me your recording ?
Also, do you have any locked orientation in manifest for this scanning activity or not ?

mtangoo, 2015-09-16:
Let me try it again.

mtangoo, 2015-09-16:
Its working now. I don't know what was wrong but I guess it have to do with installation;
Case Closed ;)

closed 2015-09-16 by mtangoo

livotov, 2015-09-16:
cool! :)