LivotovLabs/CamView #17: Add ability to switch flash on/off

mtangoo, 2015-09-16:
It will be nice to be able to switch camera flash on/off.
pseudocode
```java
scanner.start();
if(scanner.hasFlash())
    scanner.switchOn()

```

livotov, 2015-09-16:
already there :)

```
scanner.getCamera().getController().switchFlashlight(true|false);
```

mtangoo, 2015-09-16:
Let me test it ( a Thomas :))

mtangoo, 2015-09-16:
I get Null pointer exception with this code
```java
flashBtn.setOnClickListener(new View.OnClickListener() {
            @Override
            public void onClick(View v) {
                scanner.startScanner();
                if(mFlashIsOn)
                {
                    scanner.getCamera().getController().switchFlashlight(false);
                    mFlashIsOn = false;
                }
                else
                {
                    scanner.getCamera().getController().switchFlashlight(true);
                    mFlashIsOn = false;
                }

            }
        });
```

livotov, 2015-09-16:
startScanner is an async operation - it takes some time to initialize the camera. So you need to check getController() for null

livotov, 2015-09-16:
P.S. master branch already contains extended ScannerViewEventListener with extra callbacks for listening camera init/shutdown errors. This will appear in the next snapshot too.

mtangoo, 2015-09-16:
I will wait to see if it works there. So far getController() is always null

livotov, 2015-09-16:
it can't be always null, you simply trying to access it too early. See the sample app bundled + also you can now try 2.0.0.DEV4-SNAPSHOT

mtangoo, 2015-09-16:
Thanks.
I waited like 10minutes then I pressed the button to light it. is that too early? ;)
I will test the latest tomorrow. its night here!

livotov, 2015-09-16:
if you stopped the scanner at that moment - it will be back to null :)

mtangoo, 2015-09-16:
(y)

mtangoo, 2015-09-17:
tested 2.0.0.DEV4-SNAPSHOT.
It works perfect!

closed 2015-09-17 by mtangoo