LivotovLabs/CamView #18: Why is the callback for onCodeScanned() missing on "duplicated" scans?

Gira-X, 2015-10-06:
I noticed how onCodeScanned() is not called properly as expected (in the front scanner?).
Scanning 3 QR codes (in random order) with different content, say 1, 2, 3 and afterwards scanning a different code from the last one. So say: 1, 3, 2, 1

Then for the last 1 the callback is not called anymore. Is this a bug on my side? I tried lowering the timeouts to 0 and messing around with calling stopScanner() and startScanner() in the callback but I never got it to scan the last number. 

Am I doing something wrong or is this feature intentionally not supported?

livotov, 2015-10-06:
Basically no, it should not only scan the same barcode, e.g. when you present it codes 1,1,2,1 it should scan the first "1", then skip second "1" (but within the duplicate timeout value) and scan "2", then successfully scan "1" again as it only keeps the last scanned code in its memory to avoid duplicates when one performing a batch scanning.

could you please attach here all your 3 barcodes (and also name the files as 1,2,3 please), so I'll try to check this.

Gira-X, 2015-10-06:
I have been using the QR codes from http://goqr.me/ 

Just rechecked with the default startScanner(), scanning "1", "2", "3" and afterwards it will not scan "1" or "2" anymore. Could it be a problem because I use scanner.setVisibility(View.INVISIBLE) and scanner.stopScanner() in the onCodeScanned() and only call startScanner() once a  Button from the View is pressed? So I hide the view after a successful scan, but I can not explain why it would not scan the different values anymore so I am a bit puzzled here.

Gira-X, 2016-02-24:
We dynamically add the ScannerLiveView to the layout and it caused some issues where the callback was not called properly. I believe this is entirely my fault as startScanner()/stopScanner() were not called properly.

If I can reproduce it exactly, I'll create another issue.

closed 2016-02-24 by Gira-X