LivotovLabs/CamView #23: Scanning on 10" Galaxy Tab Android 4.4.2

mkallaoun, 2015-10-29:
Our 8" same android version working but not the 10". We cannot get the 10" to scan and the target icon is lost when changing vertical to horizontal. See the target icon stretching down the screen as well. 

[image]

livotov, 2015-10-30:
wow, what a weird bug. So it is not scanning (detecting) the codes at all, 
but do you see the live picture, streamed from the camera ?

Don't have the 10" Samsung tab handy, but will try to reproduce this at my side somehow.

mkallaoun, 2015-11-23:
Yea, it wasn't scanning the code at all. I just purchased a 10" so I can test it out as well. This was a client's tablet. I've only had an issue with the 10". Everything else is working great.

mkallaoun, 2015-11-27:
The mdpi camview_scanner_frame.png image was the issue. The hdpi image works fine for a 10" view. I would just copy over the hdpi image to the mdpi folder. The scan is working okay so far from what I've tested.

livotov, 2015-12-18:
oh, wow... is the original mdpi png is broken somehow ?
anyway, would appreciate the pull request with the right png which works fine on this samsung :)

kirby510, 2016-02-25:
Same thing happened to 7" tablet (Lenovo Tab 2 A7-30 HC)