LivotovLabs/CamView #25: CameraLiveView whats the use?

donpironet, 2015-12-09:
Whats the use of CameraLiveView class? Is it just to show the camera because you can't listen for QR codes or whats the difference? 

I use the ScannerLiveView good job btw.

livotov, 2015-12-09:
CameraLiveView is the base class, yes, it just shows the live stream from the camera on your activity/view without any effort. You can also grab frames and do some realtime postprocessing, if you want. This is because camera usage is not only about scanning barcodes.

ScannerLiveView is just an addon above the CameraLiveView.
It uses CameraLiveView and its frame grabbing api to perform barcodes recognition.

So if you need barcodes only - use ScannerLiveView, if you need camera for smth else - CameraLiveView would be a good starting point.

donpironet, 2015-12-09:
Oke thanks :-). I've one question though. I'm using this with the frontend camera and the scanning works but it's a bit slow vs the zxing app in the store. Do I need to do something special to just focus for QR codes or I don't know?

livotov, 2015-12-09:
yep, CamView captures entire image instead of part of it like in zxing, this may result in slower processing. Current dev branch already contains an option to define a reduced area (when necessary), but the code is not yet synced to master. Probably  I'll finish that task and sync it coming weekend, but cannot be 100% sure for now

donpironet, 2015-12-09:
aa oké :-) but I'm using it in a production app that is used by 20 000 people so I have to implement something fast. Is there no version before 2.0.0 that is fast?

livotov, 2015-12-18:
will push today some current work into the master here, with the api to change capture frame. WIll note the api usage here once pushed

livotov, 2015-12-18:
Just pushed and also updated maven with the DEV8 snapshot.
You can now reduce scanner area by providing a percent value from 0.1 (10%) to 1.0 (100%), see the bundled demo app onResume method of the MainActivity

```
@Override
    protected void onResume()
    {
        super.onResume();
        ZXDecoder decoder = new ZXDecoder();
        decoder.setScanAreaPercent(0.5);
        camera.setDecoder(decoder);
        camera.startScanner();
    }
```

donpironet, 2016-01-04:
Thank you. Is it best to only scan 50%?

livotov, 2016-01-08:
That depends on your usage scenario and on mobile devices used, I suppose.
I personally prefer to use full screen scanning.

livotov, 2016-01-08:
you can also speed up barcodes recognition by implementing custom decoder, based on ZBar sdk, as it is native code and works much faster on a full frame in comparison to zxing (at least it feels so for me)