LivotovLabs/CamView #26: Issue with startPreview() method

Rg276, 2015-12-17:
Hello Dmitri, I just ran into an issue debugging an app that I'm developing for my company. The issue is when I load a second activity (where the ScannerLiveView is), I just took a watch over the log registry and I found it was when it tries to load the camera of the device. I don't know if the API version matters because the target version of this app is 2.3.3 Gingerbread.

The log file is in the attachment.

My gradle file is like this:

apply plugin: 'com.android.application'

android {
    compileSdkVersion 23
    buildToolsVersion "23.0.2"

    defaultConfig {
        applicationId "mx.aom.ds.mantenimiento"
        minSdkVersion 10
        targetSdkVersion 23
        versionCode 1
        versionName "1.0"
        multiDexEnabled true
    }
    buildTypes {
        release {
            minifyEnabled false
            proguardFiles getDefaultProguardFile('proguard-android.txt'), 'proguard-rules.pro'
        }
    }
}
repositories {
    maven { url 'http://maven.livotovlabs.pro/content/groups/public' }
}

dependencies {
    compile fileTree(dir: 'libs', include: ['*.jar'])
    testCompile 'junit:junit:4.12'
    compile 'com.android.support:appcompat-v7:23.1.1'
    compile project(':EncriptarDesencriptar')
    compile project(':jtds-1.2.5')
    compile ('eu.livotov.labs:camview:2.0.0.DEV5-SNAPSHOT@aar') {transitive=true}
}

Also I included the right permissions on the manifest:
    <uses-permission android:name="android.permission.INTERNET" />
    <uses-permission android:name="android.permission.ACCESS_NETWORK_STATE" />
    <uses-permission android:name="android.permission.ACCESS_WIFI_STATE" />
    <uses-permission android:name="android.permission.CAMERA" />
    <uses-permission android:name="android.permission.VIBRATE"/>
    <uses-permission android:name="android.permission.FLASHLIGHT"/>

By the way, thanks you for your help, I really appreciate it.
Greetings.

[log.txt](https://github.com/LivotovLabs/CamView/files/65602/log.txt)

livotov, 2015-12-17:
Sorry, but what was the issue, does this method hangs there ?

Rg276, 2015-12-17:
Whey I try to run my application, it starts with a log in activity and then goes to the next activity (where ScannerLiveView is) and it hangs there, I just took a look on the log, and that's where it hangs, in execution time, the app only restarts but in debugging mode throws this exception.

livotov, 2015-12-18:
hmm, what OS version you use when debugging ? 

Basically, never tested the lib on 2.x as it is not supported, so I think the problem could be in the camera initialization thread, where some 2.x incompatible api call may happen. You probably need to put a breakpoint there and check.

I can check it quickly only with the 2.3 emulator this weekend, as I don't have any real 2.x device.

Rg276, 2015-12-18:
My OS is Windows 7 Professional x64 with SP1, I would appreciate it, I will keep checking if is an error of mine and trying different ways to run the scanner too.
Thanks you.

livotov, 2015-12-18:
No, I mean on what Android OS version you're experiencing this bug

Rg276, 2015-12-18:
2.3.3 Gingerbread

JStumpp, 2016-01-19:
I am having the same issue with a Huawei Ideos X3 on Android 2.3.3.

rzetzsche, 2016-01-19:
I have tested this library on a android 2.3.3 device and got the same exception.