LivotovLabs/CamView #30: ScannerLiveView doesn't fill the entire SurfaceView

papageorgiouk, 2016-01-12:
On portrait, I've set layout_width="match_parent" and layout_height=[arbitrary] (376dp in my case). The result is that the surfaceview takes the entire area, but it's empty on the sides because the camera preview is a long narrow vertical strip in the middle. 

I don't care about preserving aspect ratio, as long as the barcode is scanned.

Works fine on landscape.

livotov, 2016-01-12:
Yes, its currently preserves the ratio. 
Probably a configuration option not to do so?

papageorgiouk, 2016-01-12:
That would be nice. 

Any suggestions to temporarily work around this?

eriforce, 2016-02-08:
Looking for a workaround too

kirby510, 2016-02-24:
Actually you can just comment out some part so that it won't empty on sides...