LivotovLabs/CamView #31: Sample App Doesn't work with Android 6.0

sylvanaar, 2016-01-12:
Under the new permissions model you have to request access to the camera at runtime. The sample app doesn't do this. So in order to make it work you have to compile it on API 22 instead of API 23 for use on Android 6.0+

I am not sure if the library works under the new permissions model, as I only tested the sample app.

livotov, 2016-01-12:
Thanks, yes, update for demo app is planned.
Regarding the library - it does not deal with the permissions requesting as this is out of CAMView level. The requests/checks needs to be done at the activity level. However some helper class or similar will be added to simplify this task

Mithrandir21, 2016-03-30:
I am having the same issue with an App I'm developing.
So I was wondering if there was any news on this issue and/or if you guys could suggest a work around?

brunolanevik, 2016-04-11:
@Mithrandir21 It's all  documented [here](http://developer.android.com/training/permissions/requesting.html). It's quite simple to follow actually. I personally believe that this isn't something that should be implemented by the CamView library.