LivotovLabs/CamView #39: maven.livotovlabs.pro is not available

Disord, 2016-02-02:
Hello! I'm sorry, but maven.livotovlabs.pro is not available. When I try to put this into gradle:
repositories {
    ...
maven { url 'http://maven.livotovlabs.pro/content/groups/public' }
    ...
}

livotov, 2016-02-03:
You can download sources form the master or particular version from releases and include it as a local aar library:

- download an appropriate version of zxscanlib (or download the "master" one)
- in your project's root folder, create the folder where you will store the local libs, say "local-libs"
- copy zxscanlib/zxscanlib folder into the local-libs
- reference it in your main settings.gradle:    

```
include 'local-libs:zxscanlib'
```

- replace old compile statement in your module's build.gradle with the following:   

```
compile project(':local-libs:zxscanlib')
```

Disord, 2016-02-03:
Sorry, but can you upload jar file of **zxscanlib:2.0.1**? I've tryed and I've been noticed that gradle of zxscanlib project includes this string:
**compile 'eu.livotov.labs:camview:1.1.4-SNAPSHOT@aar'**
And it also failed to resolve. I'm trying to do what you say in this way, but there is no settings.gradle in zxscanlib folder.

livotov, 2016-02-05:
oh yeah, I forgot about old camview.. Sure, let me prepare the old package then. Will try to get it ready this weekend.

Disord, 2016-02-05:
Thank you, but I've been configured my project alredy :) I've been included camview and setted up dependencies correctly. Thank you!

GuyPaddock, 2016-04-05:
Also affected by this...

GuyPaddock, 2016-04-05:
@Disord: What did you do? How?

Disord, 2016-04-05:
I just download sources and include it by the way that **livotov** described.

GuyPaddock, 2016-04-06:
How did you get zxscanlib to resolve the reference to camview?

GuyPaddock, 2016-04-07:
@livotov: I've setup a temporary JCentral repo with your last artifacts for version 2.0.1 of zxscanlib and 1.0.3 of camview. (One of the developers on our team still had the artifacts in his Gradle cache).

The repo is here: https://bintray.com/guypaddock/LivitovLabsArchive

I'd recommend that you copy the versions over to the main LivotovLabs JCentral repo, if you can, so that there's no confusion. I tried to be clear in the descriptions on my copy that it's a legacy older version.

FireZenk, 2016-09-06:
Still not fixed...