LivotovLabs/CamView #42: Set ScannerLiveView to Only Responds to QR Code

Jayvd, 2016-02-16:
Hello, thanks for providing this CRAZY SIMPLE libray!!!

I have a little problem though.

1. How do you detect only QR Codes? I tried to scan other type of barcodes and the App BEEP. How do I make it to beep or responds only when a QR Code is scanned??

2. What is the purpose of setScanAreaPercent? (I am sorry for these newbie question)

        ZXDecoder decoder=new ZXDecoder();
        decoder.setScanAreaPercent(0.5);
        scannerLiveView = (ScannerLiveView) findViewById(R.id.scannerLiveView);
        scannerLiveView.setDecoder(decoder);

And thanks again for providing this great library for everyone!!!!

livotov, 2016-02-17:
Hi !
Detectable codes management is a planned feature for the ScannerLiveView.
Sadly I was very busy past weeks, but I hope to complete and push the outstanding issues for the new release during the coming weekend, so this one should be added too.

Will post here  more info once it ready

Jayvd, 2016-02-17:
@livotov I ended up creating a new custom decoder class derived from ZXDecoder and set the POSSIBLE_FORMATS to QR_CODE only.

Thank you for responding!!

hohl, 2016-02-23:
@Jayvd I'm just having the same issue. Do you have a short sample how you did setup ZXDecoder?

GermanMontejo, 2016-03-31:
@Jayvd Hi. Could you show us how to set the possible formats to QR code only? Please. That would be a big help for me! Thanks!

Jayvd, 2016-03-31:
@GermanMontejo  @hohl  @livotov 

What I did was, I copied the ZXDecoder.java file and name it something else. In this case I named it CustomDecoder
Change the constructor content of the ZXDecoder

    public CustomDecoder()
    {
        reader = new MultiFormatReader();

        Map<DecodeHintType,Object> hints = new EnumMap<>(DecodeHintType.class);
        Collection<BarcodeFormat> decodeFormats = EnumSet.of(BarcodeFormat.QR_CODE);
        hints.put(DecodeHintType.POSSIBLE_FORMATS, decodeFormats);

        hints.put(DecodeHintType.CHARACTER_SET, "utf-8");
        hints.put(DecodeHintType.TRY_HARDER, true);

        reader.setHints(hints);
    }

jjhesk, 2016-05-14:
it should have an option from decode to set decoding type.