LivotovLabs/CamView #45: Camera system error

vaclavpavlicek, 2016-02-24:
Hi everyone,
I have a problem with camera. I sometimes get error, but sometimes it works well.

Here is my error log.
```
02-24 17:23:27.390 19600-19600/com.vuzix.m100empty E/ScannerLiveView: n/a
02-24 17:23:27.390 19600-19600/com.vuzix.m100empty E/SCANNER: java.lang.RuntimeException: Camera system error -1
```
And there are my permissions from manifest.
`
    <uses-feature android:name="android.hardware.camera" />
    <uses-feature android:name="android.hardware.camera.autofocus" />

    <uses-permission android:name="android.permission.CAMERA" />
`
I am getting this error on Android 4.0.3. Thanks for help!

vaclavpavlicek, 2016-02-27:
I have fix it! I had to stop scanner

cbranca, 2017-09-10:
I have the same error "Camera system error -1". I see a only a black screen with the red target.
I tried your solution but it didn't worked for me.