LivotovLabs/CamView #46: Default camera picture size is the smallest one in some cases.

HugoMatilla, 2016-02-29:
Alternate title "Option to get and update default camera parameters"
Hi @livotov first,  thanks for your work on this Library.

In some devices when taking a picture, the resulting picture has really small resolution although the preview looks good.
After some searching I found that  `parameters.getPictureSize()` is different to `parameters.getPreviewSize() ` . So the preview image can look ok, but the photo taken has really low resolution.

It is because the default parameters gotten from the camera  are set to this low resolution values. They are the default parameters  [point 2 on the list](http://developer.android.com/intl/es/reference/android/hardware/Camera.html)  and I could not find how are they set in the device or where they come from. It is something that the  user set on its device?, or it depends on the device itself?  I have a list of the devices where I found this issue and there are from low to top quality devices, so I would say that is more a user thing than a device.

Is this a problem?
Is the library intention to behave this way?

If the answer is no, to fix this specific problem, *setting proper picture size when it is lower than the preview size*,  I added some methods to the `DefaultCameraV1Controller` I let them here and if you think they are ok I can make a pull request.

```java

private void updatePictureSizeIfPreviewSizeIsBigger() {
        Camera.Parameters parameters = CameraUtilsV1.getMainCameraParameters(rawCameraObject);
        if (isPreviewSizeBiggerThanPictureSize(parameters)) {
            Size betterPictureSize = getClosestSupportedPictureSizeToPreviewSize(parameters);
            parameters.setPictureSize(betterPictureSize.width, betterPictureSize.height);
            rawCameraObject.setParameters(parameters);
        }
    }

    private boolean isPreviewSizeBiggerThanPictureSize(Camera.Parameters parameters) {
        boolean widthOk = parameters.getPreviewSize().width <= parameters.getPictureSize().width;
        boolean heightOk = parameters.getPreviewSize().height <= parameters.getPictureSize().height;
        return !(widthOk && heightOk);
    }

    private Size getClosestSupportedPictureSizeToPreviewSize(Camera.Parameters parameters) {
        Size result = new Size(0, 0);
        int previewHeight = parameters.getPreviewSize().height;
        int previewWidth= parameters.getPreviewSize().width;

        List<Camera.Size> pictureSizes = parameters.getSupportedPictureSizes();
        for (Camera.Size size : pictureSizes) {
            if (size.height >= previewHeight && size.width >= previewWidth) {
                result.height = size.height;
                result.width = size.width;
            }
        }
        return result;
    }
    private static class Size {
        public  int width;
        public  int height;
        public Size(int width, int height) {
            this.width = width;
            this.height= height;
        }
    }
```
I added `updatePictureSizeIfPreviewSizeIsBigger()` at the end of the `startPreview` method before the `setupSurfaceAndCameraForPreview` call

Another solution and an enhancement is to have access to the parameters of the camera, directly from the CameraLiveView.

I hope I was clear enough :)

Thanks again for your work.

samy-baili, 2016-09-21:
Did you find a solution for this issue ?

samy-baili, 2016-09-21:
Works for me, Thank you :)