LivotovLabs/CamView #48: suddenly, no longer work ㅠㅠ

JungHaeMin, 2016-03-08:
Failed to resolve: eu.livotov.labs:zxscanlib:2.0.1

Error:A problem occurred configuring project ':app'.
> Could not resolve all dependencies for configuration ':app:_debugCompile'.
   > Could not resolve eu.livotov.labs:zxscanlib:2.0.1.
     Required by:
         SmartLG_ADMIN_LAST2:app:unspecified
      > Could not resolve eu.livotov.labs:zxscanlib:2.0.1.
         > Could not get resource 'http://maven.livotovlabs.pro/content/groups/public/eu/livotov/labs/zxscanlib/2.0.1/zxscanlib-2.0.1.pom'.
            > Could not GET 'http://maven.livotovlabs.pro/content/groups/public/eu/livotov/labs/zxscanlib/2.0.1/zxscanlib-2.0.1.pom'.
               > maven.livotovlabs.pro

help me

livotov, 2016-03-08:
zxscanlib is a deprecated project, it was merged with the CAMView.

You can either migrate to using CAMView instead of zxscanlib (it is simple, just use the new ScannerLiveView class in your layout) or grab the old sources form the corresponding repository (zxscanlib + old camview) and use them locally (as local module).

In the meantime I'll publish the old package into jcenter for legacy apps

JungHaeMin, 2016-03-08:
'O' Always thank you