LivotovLabs/CamView #57: getParameters failed (empty parameters)

anjawahl, 2016-06-07:
Hi,

sometimes i get the following error. Happened on samsung s6 with android 6.0.1. Happens only once in a while. App is switching scanner on and off again very often.

```
java.lang.RuntimeException: getParameters failed (empty parameters)
                                                                   at android.hardware.Camera.native_getParameters(Native Method)
                                                                   at android.hardware.Camera.getParameters(Camera.java:1996)
                                                                   at eu.livotov.labs.android.camview.camera.v1.DefaultCameraV1Controller.rechargePreviewBuffer(DefaultCameraV1Controller.java:169)
                                                                   at eu.livotov.labs.android.camview.camera.v1.DefaultCameraV1Controller.requestLiveData(DefaultCameraV1Controller.java:158)
                                                                   at eu.livotov.labs.android.camview.ScannerLiveView.resumeGrabbing(ScannerLiveView.java:146)
                                                                   at eu.livotov.labs.android.camview.ScannerLiveView.onReceiveProcessedCameraFrame(ScannerLiveView.java:275)
                                                                   at eu.livotov.labs.android.camview.camera.AbstractController$ProcessingResultHandler.handleMessage(AbstractController.java:43)
                                                                   at android.os.Handler.dispatchMessage(Handler.java:102)
                                                                   at android.os.Looper.loop(Looper.java:158)
                                                                   at android.app.ActivityThread.main(ActivityThread.java:7229)
                                                                   at java.lang.reflect.Method.invoke(Native Method)
                                                                   at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1230)
                                                                   at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1120)
```

Thanks !

samy-baili, 2016-07-07:
I have the same issue, @anjawahl did you find a solution ?

rohannexialabs, 2016-09-06:
I am facing same issue as well.

mrArtCore, 2017-06-26:
Seems it still haven't been fixed... Faced with same problem.